pylessard/python-can-isotp #133: There is no padding bytes when ISOTP send CANFD frame whose len is less than 7

CN-CaoJun, 2024-11-08:
Hello

After using this TP layer to send data, I found that under the current settings, if I send CANFD frames with a length less than 7 bytes, the data I see on the bus is not padded. 

params

```c
isotp_params = {
    'stmin': 10,
    'blocksize': 0,
    'override_receiver_stmin': None,
    'wftmax': 4,
    'tx_data_length': 64,
    'tx_padding': 0x00,
    'rx_flowcontrol_timeout': 1000,
    'rx_consecutive_frame_timeout': 100,
    'can_fd': True,
    'max_frame_size': 4095,
    'bitrate_switch': False,
    'rate_limit_enable': False,
    'listen_mode': False,
    'blocking_send': False    
}

isotp_layer = isotp.NotifierBasedCanStack(
    bus = bus,
    notifier = notifier,
    address = tp_addr,
    error_handler = None,
    params = isotp_params
)
```

**Case One** -> If  isotp_layer send `22 F1 86` 

Expected Data

```
0x03 0x22 0xF1 0x86 0x00 0x00 0x00 0x00
```

Actual Data

```
0x03 0x22 0xF1 0x86 
```

**CASE Two** -> If isotp_layer receive multi CANFD frames, isotp_layer send FC is less than 8 bytes, so that the sender could not recognize the response frame.

Expected Data

```
30 00 10 00 00 00 00 00
```

Actual Data

```
30 00 10
```

**Small fix** -> In `_get_nearest_can_fd_size` func,  it should return 8 when sized <=8

```c
 def _get_nearest_can_fd_size(self, size: int) -> int:
        if size <= 8:
            return size 
```

```c
 def _get_nearest_can_fd_size(self, size: int) -> int:
        if size <= 8:
            return 8 
```

After thix change , isotp_layer  could send at least 8 bytes frame.

hartkopp, 2024-11-08:
> **Case One** -> If isotp_layer send `22 F1 86`
> 
> Expected Data
> 
> ```
> 0x03 0x22 0xF1 0x86 0x00 0x00 0x00 0x00
> ```
> 
> Actual Data
> 
> ```
> 0x03 0x22 0xF1 0x86 
> ```
> 

How did you get the "Actual Data"?

If you read an ISO-TP PDU you won't see the padding. Which is the correct behavior.

E.g.

send a CAN FD ISO-TP PDU with padding "CC" of length 4:
`$ isotpsend vcan0 -s 321 -d 123 -L 72:64:1 -D 4 -p CC:DD`

With isotprecv you get the ISO-TP PDU content

```
$ isotprecv -s 123 -d 321 -l -L 72:64:1 vcan0
01 02 03 04 
```
With candump you see the raw CAN FD frame

```
$ candump any
  vcan0  321  [08]  04 01 02 03 04 CC CC CC
```

pylessard, 2024-11-09:
What you want is setting 'tx_min_data_length' to enforce padding.
See : https://can-isotp.readthedocs.io/en/latest/isotp/implementation.html#tx_data_min_length

CN-CaoJun, 2024-11-11:
> How did you get the "Actual Data"?
>
> If you read an ISO-TP PDU you won't see the padding. Which is the correct behavior.
>
> E.g.
>
> send a CAN FD ISO-TP PDU with padding "CC" of length 4: `$ isotpsend [...]

Hello, thank you very much for your comment
I conducted the test using Vector hardware and with FD enabled. As the author stated, without adding the parameter 'tx_data_inlength': 8, when calling the send method, the remaining bytes cannot be padded with padding values.

Here is the test result,  when call

```
payload = bytearray([x & 0xFF for x in range(int(user_input))])
...
isotp_layer.send(payload)
```
Result:
```shell
DEBUG:isotp:Given rxfn is considered blocking
DEBUG:isotp:Starting NotifierBasedCanStack
DEBUG:isotp:Main thread has started
DEBUG:isotp:Relay thread has started
Please enter len data or the hexadecimal data to be sent (e.g. 1A2B3C), or enter 'q' to exit:4
DEBUG:isotp:Enqueuing a SendRequest for 4 bytes and TAT=Physical
Send Request, Len =  4
DEBUG:isotp:Tx: <7E1> (05) [ ]   0400010203
```
[image]

CN-CaoJun, 2024-11-11:
> What you want is setting `tx_min_data_length` to enforce padding. See : https://can-isotp.readthedocs.io/en/latest/isotp/implementation.html#tx_data_min_length

Hello, this setting can take effect. Thank you very much!

closed 2024-11-11 by CN-CaoJun