pylessard/python-can-isotp #134: One second delay occurs between consecutive frames.

HyoungSikKWON, 2024-11-11:
First of all, thank you very much for creating such a useful module.

When using CAN-FD, a delay of 1 second occurs between consecutive frames as shown below.

[image]

pylessard, 2024-11-11:
You set ``override_receiver_stmin=1``   You are enforcing a delay of 1 second.
https://can-isotp.readthedocs.io/en/latest/isotp/implementation.html#override_receiver_stmin

I realize that there is a discrepency between ``stmin`` and ``override_received_stmin``. On takes byte value that maps to a time, the second takes a time value given as float.

HyoungSikKWON, 2024-11-11:
Sorry for wasting your precious time. I understood it to be the same as the "stmin" parameter, not seconds. thank you. 
I will read the document carefully from now on.

pylessard, 2024-11-11:
No worries. I would have assumed the same.
Passing a time values gives more freedom and does not constrain someone to use values available in iso15765

closed 2024-11-11 by pylessard