pylessard/python-can-isotp #138: Getting P2 Timeout when sending uds request and getting response to vn1640A

Nishanth-cts, 2025-03-21:
Hello Pylessard I am using ur isotp connection to send uds request to vn1640a but I end up with a p2 timeout error. Can you check my code and let me know where I am going wrong?

pylessard, 2025-03-21:
ok, but you should share your code.
Also, a CAN log would be even better

Nishanth-cts, 2025-03-21:
I am not able to attach the file here. Can you provide me ur email id so that I can send the file

Nishanth-cts, 2025-03-21:
Sorry Pylessard as I am new to usage of git i am nit aware of the flow process anyway as you suggested I will execute my scripts and share some screenshots. I totally understand concern. Thanks. Will get attach screenshots of my code here in sometime.

pylessard, 2025-03-21:
Just copy paste your stack configuration and and an example of how you call the UDS client. I don't want your whole program

and, again. A Can log is the most important thing for debugging here.

Nishanth-cts, 2025-03-21:
Hi Lessard, due to my company policies i was not able to get the can log but i have screenshots as well as the whole code I am sharing the code snippets.

[image]
[image]
[image]

Nishanth-cts, 2025-03-21:
Also, for your information I am sending continuous can messages to wake up the ecu while i am sending this UDS request not sure if that is causing this timeout issue.

Nishanth-cts, 2025-03-21:
[image]

Nishanth-cts, 2025-03-21:
[image]

Nishanth-cts, 2025-03-22:
Hello Pylessard any way you can suggest in order to handle the above scenario?

pylessard, 2025-03-23:
No, do some basic debugging first.
Check at your can communication. see if messges are going out, if they do, check if the server respond.

Nishanth-cts, 2025-03-23:
Can communication happens but I get a global timeout from the server.

pylessard, 2025-03-23:
Ok

closed 2025-03-23 by pylessard