pylessard/python-can-isotp #139: Getting dlc not set to 8 as expected when using client.session(1)

Nishanth-cts, 2025-03-24:
Hello Pylessard, after setting isotp parameters for vn1640a, and trying to enter into default session using client.session(1) in my canalyzer trace I see I get dlc is not set to 8 as expected and hence response to my 10 01 request I am unable to see. Could you kindly let me know where I am going wrong?

Nishanth-cts, 2025-03-24:
[image]

Nishanth-cts, 2025-03-24:
[image]

Nishanth-cts, 2025-03-24:
Attached above are the isotp, client configuration as well as uds request sent.

pylessard, 2025-03-24:
@Nishanth-cts 
First of all stop sending pictures of your screen. Just copy paste the relevant piece of code.

Second, DLC of 8 is not required. If you want a dlc of 8 bytes, you can set a ``tx_data_min_length`` parameter.

Also, you are making very reckless usage of threads. Running a thread that reads you can box may deplete the internal buffer and stop messages from reaching the isotp stack. You can add a listener that prints instead of using a thread.

You should also read about threading.Event object. Instead of using a global variable and doing sleep(1). You can just do event.wait() in your isotp thread and event.set() in your tkinter thread.  It will react in microseconds and avoid bloating your cpu.

pylessard, 2025-03-26:
I wrote an extensive documentation. Please read it 

https://udsoncan.readthedocs.io/en/latest/udsoncan/examples.html#reading-a-did-with-readdatabyidentifier
https://udsoncan.readthedocs.io/en/latest/udsoncan/client.html#udsoncan.client.Client.read_data_by_identifier

Nishanth-cts, 2025-03-27:
Hello Pylessard, 
Thanks very much it works. I just have one last question. When i try to First enter into default session using client.change_session(1) followed by client.change_session(3) I only see response for 10 01 but not for 10 03. 
Could you kindly let me know how to handle such scenarios where consecutive UDS requests have to be sent.

[image]
[image]